huayan-robotics/elfin_robot #5: Ethercat communication details

felix-bo-hu, 2018-04-16:
Hello,

I noticed there were lots of definitions related with ethercat rx/txpdo. For example:
```
namespace elfin_txpdo
{
    const int AXIS1_STATUSWORD=0;
    const int AXIS1_ACTPOSITION=1;
    const int AXIS1_ACTCUR=2;
    const int AXIS1_ERRORCODE=3;

    const int AXIS2_STATUSWORD=4;
    const int AXIS2_ACTPOSITION=5;
    const int AXIS2_ACTCUR=6;
    const int AXIS2_ERRORCODE=7;

    const int UDM_STATUS=8;
    const int ACCELERATION_X=9;
    const int ACCELERATION_Y=10;
    const int ACCELERATION_Z=11;
}

namespace elfin_rxpdo
{
    const int AXIS1_CONTROLWORD=0;
    const int AXIS1_TARGET_POSITION=1;
    const int AXIS1_ENDATPOS_FLASH=2;
    const int AXIS1_FEEDFORWARD_CUR=3;

    const int AXIS2_CONTROLWORD=4;
    const int AXIS2_TARGET_POSITION=5;
    const int AXIS2_ENDATPOS_FLASH=6;
    const int AXIS2_FEEDFORWARD_CUR=7;

    const int UDM_CMD=8;
}

```
I want to know where are those definitions coming from? Because I couldn't find any ESI file of the slave.

onionsflying, 2018-04-16:
Hi,
There is a special ethercat protocol for elfin. You can ask the [vendor](http://en.hansrobot.com/) for a document about the protocol. By the way, there is a binary ESI file in [elfin_ethercat_driver/script](https://github.com/hans-robot/elfin_robot/tree/master/elfin_ethercat_driver/script) for [Fix ESI](https://github.com/hans-robot/elfin_robot/blob/master/docs/Fix_ESI_english.md).

ahcorde, 2018-10-25:
I'm trying to test the module example with a real hans module. But I'm getting the following error:

```
SOEM found and configured 1 slaves
Could not set EC_STATE_PRE_OP
terminate called after throwing an instance of 'elfin_ethercat_driver::EtherCatError'
  what():  Could not initialize SOEM
```

I have fixed the ESI file of the motor with the following [documentation](https://github.com/hans-robot/elfin_robot/blob/master/docs/Fix_ESI_english.md). But the code is not able to change the state to PRE_OP.

Any thoughts?

onionsflying, 2018-10-26:
Hi, @ahcorde 

Your robot seems to have the version 2 of elfin EtherCAT slaves. Please update the elfin_robot package first and fix the ESI with elfin_motor_v2.esi. Then try *elfin_ros_control_v2.launch* instead of *elfin_ros_control.launch*.

knez93, 2018-10-26:
I load this FIX and now I can control robot through ROS. But my robot dot work whit teach pendant. It said that Master start is Failed.

knez93, 2018-10-26:
The Error is topological unmatched ENI file .....

onionsflying, 2018-10-26:
Yes, you can't control the robot with ROS and teach pendant at the same time. Because the computer in the control box was disconnected from the robot when you connected the robot to ROS.

knez93, 2018-10-26:
I returned Ethernet cable from the control box to etherCAT driver. Do you have some new version of .xml file that is compatible whit this elfin_motor_v2.esi file

onionsflying, 2018-10-26:
If you've got one .xml file, it must be compatible with the elfin_motor_v2.esi file. If you haven't, please ask my colleague who sold you the robot. I only have the binary files.

knez93, 2018-10-26:
Can you pleas send me the files that you have?

onionsflying, 2018-10-27:
I mean the files that I have are already in this repository: elfin_motor.esi and elfin_motor_v2.esi.

closed 2018-11-02 by onionsflying