huayan-robotics/elfin_robot #6: failed runing "roslaunch elfin_basic_api elfin_basic_api.launch"

chouer19, 2018-07-04:
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://XueChong:39545/

SUMMARY
========

PARAMETERS
 * /elfin_basic_api/octomap_resolution: 0.1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    elfin_basic_api (elfin_basic_api/elfin_basic_api_node)
    elfin_gui (elfin_basic_api/elfin_gui.py)

ROS_MASTER_URI=http://localhost:11311

process[elfin_basic_api-1]: started with pid [27597]
process[elfin_gui-2]: started with pid [27598]
[ERROR] [1530688192.231223147]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1530688192.231346224]: Loading robot model 'elfin3'...
[ INFO] [1530688192.231382531]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1530688192.410539093]: Group 'elfin_arm' was not found.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Group 'elfin_arm' was not found.
[elfin_basic_api-1] process has died [pid 27597, exit code -6, cmd /home/chouer/workspace/rospace/devel/lib/elfin_basic_api/elfin_basic_api_node __name:=elfin_basic_api __log:=/home/chouer/.ros/log/37ba34d0-7f54-11e8-8242-708bcd827eeb/elfin_basic_api-1.log].
log file: /home/chouer/.ros/log/37ba34d0-7f54-11e8-8242-708bcd827eeb/elfin_basic_api-1*.log
Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
[ERROR] [1530688193.043079847]: Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?
[ INFO] [1530688193.043148851]: Loading robot model 'elfin3'...
[ INFO] [1530688193.043168720]: No root/virtual joint specified in SRDF. Assuming fixed joint
[FATAL] [1530688193.239256938]: Group 'elfin_arm' was not found.
Traceback (most recent call last):
  File "/home/chouer/workspace/rospace/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py", line 630, in <module>
    myframe=MyFrame(parent=None,id=-1)  
  File "/home/chouer/workspace/rospace/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py", line 73, in __init__
    self.group=moveit_commander.MoveGroupCommander('elfin_arm')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Group 'elfin_arm' was not found.
[elfin_gui-2] process has died [pid 27598, exit code 1, cmd /home/chouer/workspace/rospace/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py __name:=elfin_gui __log:=/home/chouer/.ros/log/37ba34d0-7f54-11e8-8242-708bcd827eeb/elfin_gui-2.log].
log file: /home/chouer/.ros/log/37ba34d0-7f54-11e8-8242-708bcd827eeb/elfin_gui-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

===============================================================================
my machine environment is ubuntu16.04 + ros kinetic
and I update moveIt using  "sudo apt update & sudo apt-get install ros-kinetic-moveit"

onionsflying, 2018-07-04:
The following launch files should remain running status when you run "roslaunch elfin_basic_api elfin_basic_api.launch".

```sh
$ roslaunch elfin_gazebo elfin3_empty_world.launch
```
```sh
$ roslaunch elfin3_moveit_config moveit_planning_execution.launch
```

chouer19, 2018-07-04:
@onionsflying yes, I did. But failed like above.

and now there are some new errors:

SUMMARY
========

PARAMETERS
 * /elfin_basic_api/octomap_resolution: 0.1
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    elfin_basic_api (elfin_basic_api/elfin_basic_api_node)
    elfin_gui (elfin_basic_api/elfin_gui.py)

ROS_MASTER_URI=http://localhost:11311

process[elfin_basic_api-1]: started with pid [16214]
process[elfin_gui-2]: started with pid [16215]
**### [ERROR] [1530695467.637006398]: Semantic description is not specified for the same robot as the URDF**
[ INFO] [1530695467.637270007]: Loading robot model 'elfin5'...
[ INFO] [1530695467.637303730]: No root/virtual joint specified in SRDF. Assuming fixed joint
### **[ERROR] [1530695468.352511620]: Semantic description is not specified for the same robot as the URDF**
[ INFO] [1530695468.352682159]: Loading robot model 'elfin5'...
[ INFO] [1530695468.352700859]: No root/virtual joint specified in SRDF. Assuming fixed joint
**### Traceback (most recent call last):
  File "/home/chouer/workspace/rospace/elfin/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py", line 630, in <module>
    myframe=MyFrame(parent=None,id=-1)  
  File "/home/chouer/workspace/rospace/elfin/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py", line 73, in __init__
    self.group=moveit_commander.MoveGroupCommander('elfin_arm')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 52, in __init__
    self._g = _moveit_move_group_interface.MoveGroupInterface(name, robot_description, ns)
RuntimeError: Unable to connect to move_group action server 'move_group' within allotted time (5s)
[elfin_gui-2] process has died [pid 16215, exit code 1, cmd /home/chouer/workspace/rospace/elfin/src/elfin_robot-kinetic-devel/elfin_basic_api/src/elfin_gui.py __name:=elfin_gui __log:=/home/chouer/.ros/log/7cfc5590-7f5f-11e8-8242-708bcd827eeb/elfin_gui-2.log].
log file: /home/chouer/.ros/log/7cfc5590-7f5f-11e8-8242-708bcd827eeb/elfin_gui-2*.log
^C[elfin_basic_api-1] killing on exit
^Cterminate called after throwing an instance of 'std::runtime_error'**
  what():  Unable to connect to move_group action server 'move_group' within allotted time (0s)
shutting down processing monitor...
... shutting down processing monitor complete
done

chouer19, 2018-07-04:
@onionsflying 
roslaunch elfin_gazebo elfin5_empty_world.launch
and
roslaunch elfin_basic_api elfin_basic_api.launch
do not work, and errors occurred

=======================

but 
roslaunch elfin3_moveit_config moveit_planning_execution.launch
and 
roslaunch elfin_basic_api elfin_basic_api.launch
worked well

chouer19, 2018-07-04:
@onionsflying 
I'm sorry!
I should rosrun all three launch files!
Thanks for your contribution of this simulation!

closed 2018-11-02 by onionsflying

Daniel-CUHK, 2022-02-11:
@chouer19 Hi, I meet the same problem with "Semantic description is not specified for the same robot as the URDF", how to solve it?