huayan-robotics/elfin_robot #7: elfin_ethercat_driver::EtherCatError

Leytha, 2018-07-04:
Hi, I have an Elfin 5 robot, and I'm trying to control it through ROS, on Ubuntu 16.04 and ROS Kinetic.
First, I tried with the simulator, and everything works correct if I run these three:

_roslaunch elfin_gazebo elfin5_empty_world.launch
roslaunch elfin5_moveit_config moveit_planning_execution.launch display:=false
roslaunch elfin_basic_api elfin_basic_api.launch_

Then, I'm trying to connect with my real robot, already electrified, whose SW version is 2.5.263.266. The robot is connected using a LAN cable to my compute. I manually set my computer's IP to 192.168.0.30, and the robot's one is 192.168.0.25. I've already sent TCP/IP string messages to this IP and it worked, and when I ping the robot from the computer and vice versa, it's working. 
However, the robot has another connexion, I guess it's with the EtherCat servos, whose IP is 192.168.0.20.
Running ifconfig in my computer gives me the name of the connexion with the robot, which is enp3s0, so in my elfin_robot/elfin_robot_bringup/config/elfin_drivers.yaml file its written _elfin_ethernet_name: enp3s0_. I got the file from the manufacturer and I didn't change anything apart from that. 

Then I run:
_roslaunch elfin_robot_bringup elfin5_bringup.launch_
and
_sudo chrt 10 bash_
_roslaunch elfin_robot_bringup elfin_ros_control.launch_
 in two different shells. The first one doesn't return any errors, but the second one returns the following:

----------------------------------------------------------------------------------------------------------------------------------
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://GL553VD:35653/

SUMMARY

PARAMETERS
 * /elfin_arm_controller/action_monitor_rate: 10
 * /elfin_arm_controller/constraints/goal_time: 0.6
 * /elfin_arm_controller/constraints/stopped_velocity_tolerance: 0.1
 * /elfin_arm_controller/joints: ['elfin_joint1', ...
 * /elfin_arm_controller/state_publish_rate: 25
 * /elfin_arm_controller/stop_trajectory_duration: 0.05
 * /elfin_arm_controller/type: position_controll...
 * /elfin_ros_control/elfin/count_zeros: [4586774, 9794018...
 * /elfin_ros_control/elfin/joint_names: ['elfin_joint2', ...
 * /elfin_ros_control/elfin/reduction_ratios: [101, 101, 101, 1...
 * /elfin_ros_control/elfin/slave_no: [1, 2, 3]
 * /elfin_ros_control/elfin_ethernet_name: enp3s0
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    elfin_arm_controller_spawner (controller_manager/controller_manager)
    elfin_joint_controller_spawner (controller_manager/controller_manager)
    elfin_ros_control (elfin_ros_control/elfin_hardware_interface)

ROS_MASTER_URI=http://localhost:11311

process[elfin_ros_control-1]: started with pid [26009]
process[elfin_joint_controller_spawner-2]: started with pid [26010]
process[elfin_arm_controller_spawner-3]: started with pid [26011]
Initializing etherCAT master
wkc = -1
No slaves are found on enp3s0
terminate called after throwing an instance of 'elfin_ethercat_driver::EtherCatError'
  what():  Could not initialize SOEM
[elfin_ros_control-1] process has died [pid 26009, exit code -6, cmd /home/leytha/Workspaces/RoboWare/devel/lib/elfi
n_ros_control/elfin_hardware_interface __name:=elfin_ros_control __log:=/home/leytha/.ros/log/b04b2dce-7f6c-11e8-ad7
c-4d48c0de838e/elfin_ros_control-1.log].
log file: /home/leytha/.ros/log/b04b2dce-7f6c-11e8-ad7c-4d48c0de838e/elfin_ros_control-1*.log

----------------------------------------------------------------------------------------------------------------------------------

I've tried executing before _sudo su --preserve-environment_ (I got it from an answer for a similar error somebody had with a robotiq gripper https://answers.ros.org/question/276328/could-not-initialize-ethercat-driver-robotiq-3-finger-gripper/) but the answer is the same.

I guess I'm configuring wrong the connexion, but I don't know how I should configure it, or what may be causing this error. Maybe it is related with the PREEMPT_RT issue linked in the tutorial (https://wiki.linuxfoundation.org/realtime/documentation/howto/applications/preemptrt_setup), but I've searched and there is no rt version available for my linux kernel (_uname -r_ returns _4.13.0-43-generic_) , as these are the available ones : https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/.
Should I install other 4.13 rt linux kernel, although is not the one I have now? I don't want to spoil Ubuntu so I have to reinstall everything again, so I want to be sure of what I am doing. And maybe this is not the problem, as it doesn't even find the EtherCat slaves.

Thanks for your help in advance.

onionsflying, 2018-07-05:
> Should I install other 4.13 rt linux kernel, although is not the one I have now?

Yes, you can install a linux kernel that is not the same as the current one.

Leytha, 2018-07-05:
> Hi, the Elfin should be connected directly to a computer, without the control box. In this case, you need such an adapter: Elfin cable adapter.pdf

Ah, I didn't know that. I'll look for the adapter and try again.

Thanks for the help 😄

closed 2018-07-05 by Leytha

mfkenson, 2018-12-02:
> Hi, the Elfin should be connected directly to a computer, without the control box. In this case, you need such an adapter: [Elfin cable adapter.pdf](https://github.com/hans-robot/elfin_robot/files/2164590/Elfin.cable.adapter.pdf)
> 
> By the way, there is no need to set IP address for EtherCAT connection.

where could I purchase the DC power supply, and the adapter?

onionsflying, 2018-12-03:
Hi @mfkenson, you can only purchase the adapter from our company because it's not a standard element. The DC power supply we recommend is Mean Well RSP-1000-48.

jin-technoaccord, 2021-07-27:
@Leytha were you able to fix it?  I am receiving same error message.

mfkenson, 2021-07-29:
@jin-technoaccord did you use this adapter? Mine is fine with this adapter. Melodic with low latency kernel

[image]

jin-technoaccord, 2021-07-29:
@mfkenson Yes, its working now. Thanks!

jin-technoaccord, 2021-07-29:
@onionsflying 

> 
> 
> > Should I install other 4.13 rt linux kernel, although is not the one I have now?
> 
> Yes, you can install a linux kernel that is not the same as the current one.

Im using kernel version 4.15.0 but I cant find it from the list. Should I use 4.14, 4.16 or it doesnt matter?

mfkenson, 2021-07-30:
> @onionsflying
> 
> > > Should I install other 4.13 rt linux kernel, although is not the one I have now?
> > 
> > 
> > Yes, you can install a linux kernel that is not the same as the current one.
> 
> Im using kernel version 4.15.0 but I cant find it from the list. Should I use 4.14, 4.16 or it doesnt matter?

Kernel version does not really matter though realtime kernel is highly recommended. I instead use low-latency kernel for simplicity.

jin-technoaccord, 2021-07-30:
Hi @oionson, I appreciate your response. I just updated my linux to low-latency kernel. Are there significant performance difference between RealTime vs low-latency for the robotics application?
I will need to keep feeding the next position the robot needs to move in real time.

mfkenson, 2021-07-31:
> Hi @oionson, I appreciate your response. I just updated my linux to low-latency kernel. Are there significant performance difference between RealTime vs low-latency for the robotics application?
> I will need to keep feeding the next position the robot needs to move in real time.

I would say my personal experience is that both rt kernel and low-latency kernel just work fine. no gurantee