huayan-robotics/elfin_robot #8: can not run two teleop joint cmd at one time

chouer19, 2018-07-11:
when I run "rostopic pub /elfin_teleop_joint_cmd_no_limit std_msgs/Int64 "data: -1" -r 10" and "rostopic pub /elfin_teleop_joint_cmd_no_limit std_msgs/Int64 "data: -4" -r 20" in shell, some errors occurred.

at this running window "roslaunch elfin_gazebo elfin10_empty_world.launch"
==================================================================================
[joint_state_controller_spawner-6] process has finished cleanly
log file: /home/chouer/.ros/log/220304c4-84d5-11e8-a6bc-708bcd827eeb/joint_state_controller_spawner-6*.log
[elfin_arm_controller_spawner-7] process has finished cleanly
log file: /home/chouer/.ros/log/220304c4-84d5-11e8-a6bc-708bcd827eeb/elfin_arm_controller_spawner-7*.log

Segmentation fault
[gazebo-1] process has died [pid 5393, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/home/chouer/.ros/log/220304c4-84d5-11e8-a6bc-708bcd827eeb/gazebo-1.log].
log file: /home/chouer/.ros/log/220304c4-84d5-11e8-a6bc-708bcd827eeb/gazebo-1*.log

onionsflying, 2018-07-11:
elfin_basic_api has only one thread. You'd better send one command at one time.

chouer19, 2018-07-12:
@onionsflying OK,thank you very much!
What should I do, if I want to do some complex actions by controlling more than one joints on time?
I need your help.

onionsflying, 2018-07-13:
You can try other commands like "elfin_basic_api/joint_goal". The standard commands for teleoperation are two service "elfin_basic_api/joint_teleop" and "elfin_basic_api/cart_teleop" instead of "elfin_teleop_joint_cmd_no_limit". elfin_basic_api supports only these two teleoperation modes. If you want to teleoperate Elfin in other ways, you may need to write some other nodes.

chouer19, 2018-07-13:
@onionsflying 
thanks for your answer.
I will have a try using service "elfin_basic_api/joint_teleop".
And I wish I could request two or more different services at the same time.

onionsflying, 2018-07-13:
@chouer19 That will not work. Multiple joints can move at one time (When you use "elfin_basic_api/cart_teleop", you should have seen it). But I think the two existing teleoperation modes may not  meet your needs. In this case, you can write some nodes you need by yourself.

chouer19, 2018-07-16:
@onionsflying why do you think that "two existing teleoperation modes may not meet your needs"?
If so, I have a question about actions like following a trajectory. You control just one teleop one time when you make it do some complex actions? That means you make one teleoperation for one time step and then another teleoperation for another teleoperation and then and then ...... ?
I do not know how to make it when I need more complex actions. I wish I could control it with some state-of-art AI algorithms, and I think multiple teleopertions may be neccesary one day.
I need your help, please.

onionsflying, 2018-07-17:
If you wanna plan a trajectory with MoveIt! and then follow it, you can use "elfin_basic_api/joint_goal" or "elfin_basic_api/cart_goal". If you have your own trajectories, you can use "elfin_arm_controller/command". These topics are introduced in [docs/API_description_english.md](https://github.com/hans-robot/elfin_robot/blob/kinetic-devel/docs/API_description_english.md). Following a trajectory has nothing to do with teleoperation.

closed 2018-11-02 by onionsflying