huayan-robotics/elfin_robot #15: Error running the code in a real robot

LanderU, 2019-02-14:
Hello!

I'm having some issues when I try to run the code for the real robot. Here the trace:

```bash
wkc = 1
SOEM found and configured 4 slaves
SOEM IOMap size: 412

Finished configuration successfully

 recognizing joint positions, please wait a few minutes ... ... 
[ INFO] [1550148237.483144892]: positions are recognized automatically
Loaded joint_state_controller
[ERROR] [1550148237.691717503]: Could not find joint 'elfin_joint3' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1550148237.692016134]: Failed to initialize the controller
[ERROR] [1550148237.692156989]: Initializing controller 'elfin_arm_controller' failed
```

I'm using the instructions available on the README.md.

Any hints would be really helpful.

Thanks in advance for your support!

LanderU, 2019-02-14:
Solved after some minimal changes(file names)!

closed 2019-02-14 by LanderU

jin-technoaccord, 2021-07-29:
@LanderU Hi I received the similar error when I was following the elfin module tutorial
https://github.com/hans-robot/elfin_robot/blob/melodic/docs/elfin_module_tutorial_english.md
$ sudo chrt 10 bash
$ roslaunch elfin_robot_bringup elfin_module_ros_control.launch 
returns error like this
 file:///home/jin-technoaccord/Pictures/Screenshot%20from%202021-07-29%2015-00-44.png
Which files did you change the name to fix the issue ?

LanderU, 2021-07-30:
Hey @jin-technoaccord, I cannot remember the change I did sorry. If you paste correctly the error I can try to help you.

jin-technoaccord, 2021-07-30:
Hi @LanderU, sorry about the picture.
The followings are error message from 
https://github.com/hans-robot/elfin_robot/blob/melodic/docs/elfin_module_tutorial_english.md

**Finished configuration successfully

 recognizing joint positions, please wait a few minutes ... ... 
The voltage of slave 2 is: 48.494710V.
[ WARN] [1627655432.536167754]: recognizePose phase1 failed while pose recognition in slave 2, channel 1
The voltage of slave 3 is: 48.731993V.
[ WARN] [1627655432.546879589]: recognizePose phase1 failed while pose recognition in slave 3, channel 1
The voltage of slave 1 is: 48.865465V.
[ WARN] [1627655432.558699392]: recognizePose phase1 failed while pose recognition in slave 1, channel 1
[ INFO] [1627655432.567185867]: positions are recognized automatically
[ERROR] [1627655432.810065535]: Could not find joint 'elfin_module_joint1' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1627655432.810131974]: Failed to initialize the controller
[ERROR] [1627655432.810158355]: Initializing controller 'elfin_module_controller' failed
Error when loading 'elfin_module_controller'
Loaded 'joint_state_controller'**