huayan-robotics/elfin_robot #16: Error when trying to enable servos on Elfin

igorrecioh, 2019-02-14:
Hi!
- I have been following the steps provided [here](https://github.com/hans-robot/elfin_robot/blob/kinetic-devel/README_english.md).

- I am able to see actual position in RViz but when I try to Servo ON from the Elfin Control Panel I get this errors:
```bash
[ WARN] [1550159223.426480973]: setEnable phase2 in slave 3 failed
[ WARN] [1550159223.426842752]: setEnable phase2 in slave 2 failed
[ WARN] [1550159223.427466899]: setEnable phase2 in slave 1 failed
[ERROR] [1550159243.537458192]: can't start elfin_arm_controller, because module[0] is not enabled
[ERROR] [1550159243.537593708]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
```

Not sure what is going on. Any help will be useful.
Thanks in advance!

onionsflying, 2019-02-15:
That's a little weird. Could you show me the TxPDOs before and after "Servo ON"? You can get TxPDOs using the service `elfin_ros_control/elfin/get_txpdo`. Besides, you can get more information about services in [this document](https://github.com/hans-robot/elfin_robot/blob/kinetic-devel/docs/API_description_english.md#services).

igorrecioh, 2019-02-15:
Hi @onionsflying ! Thanks for your answer. Find attached a gist with the results before and after "Servo ON":
https://gist.github.com/igorrecioh/43de2f43307bd7f99b5755b26be8af71

onionsflying, 2019-02-15:
Hi, the reason of the error is that the voltage is too low. Please note that the nominal voltage is 48V DC.

igorrecioh, 2019-02-15:
Hi @onionsflying! You were right, we had some issues with our power supply. Now Elfin is working fine.

Thanks for your support!

closed 2019-02-15 by igorrecioh