huayan-robotics/elfin_robot #21: How can I use the elfin5 robot without PREEMPT_RT patch?

BobbyBelieve, 2019-10-10:
I want to keep the deep learning environment supported by NVIDIA driver. 

But the additional PREEMPT_RT patch installed in my PC that destroyed my deep learning  environment and I can't use NVIDIA driver now. 

But it seems that I can't start the controllers if I don't patch the Linux kernel.

So, How can I control the elfin5 robot without PREEMPT_RT patch?

BobbyBelieve, 2019-10-17:
Ok, I found that I could control the elfin5 robot even without the PREEMPT_RT patch.

closed 2019-10-17 by BobbyBelieve

feifeizhang8319, 2022-07-12:
> Ok, I found that I could control the elfin5 robot even without the PREEMPT_RT patch.

hello，friend，I also meet this problem，it's holy shit！ could you tell me how did you figure it out ？how could controll the elfin robot without the patch ? thank you a lot !

awupereda, 2022-10-28:
Hi, can you tell us what kernel version are you using? I have Ubuntu 18.04 and kernel 5.4 and haven't been able to connect to it