huayan-robotics/elfin_robot #36: RS485 api setup

jwwood3, 2023-06-19:
I have an Elfin5 arm and I'm trying to use the RS 485 communication ports on the end of arm connector, but I can't figure out how to program it to send and receive data in ros. The api documentation doesn't seem to have anything for this, though the readme mentions a separate v3 launch file that doesn't seem to exist anymore. I've been using the ros python libraries with ros noetic on an Ubuntu 20.04 virtual machine.

Neverforgetlove, 2023-06-20:
If you use RS485, then use the standard branch noetic.
You can control terminal IO through this service: elfin_ros_control/elfin/io_port1/write_do (elfin_robot_msgs/ElfinIODWrite)
and you can find it in docs/API_description.md
You can refer the img to control the DO output, it only have three DO
[image]

jwwood3, 2023-06-20:
Thank you for the quick response. This is the pinout I have for the end of arm connector. 
[image]

It shows the three DO ports as pins 4,5, and 6. Then pins 7 and 8 are the RS 485 connections. Can I access those ports?

Neverforgetlove, 2023-06-20:
You can only control the DO ports as pins 4,5, and 6, and do not support to control rs485

closed 2023-07-05 by jwwood3