huayan-robotics/elfin_robot #37: Collision Detection limits

jwwood3, 2023-07-05:
Hello,
where are the collision detection force cutoffs defined? I would like to be able to increase the sensitivity to collisions for safety, so even a light collision will trigger the fault.
It would also be helpful to have access to the torque sensors in the arm. Is there a way to get those readings through the api?
Thank you.

Neverforgetlove, 2023-07-06:
It dosen't provided collision detection function when you use ROS to control the robot.But you can get the joint effort in topic /joint_states
[image]

jwwood3, 2023-07-06:
> It dosen't provided collision detection function when you use ROS to control the robot.

Really? It has been stopping itself if it runs into something. Is there something else that would be doing that?

Neverforgetlove, 2023-07-07:
Only can use joint effort to judged it now