huayan-robotics/elfin_robot #39: The following error is received when running the command "roslaunch elfin_robot_bringup elfin_ros_control.launch"

oguzhanaknc, 2023-08-04:
When the command "roslaunch elfin_robot_bringup elfin_ros_control.launch" is executed, the following error is received:

```
Could not initialize ethercat driver
terminate called after throwing an instance of 'elfin_ethercat_driver::EtherCatError'
what(): Could not initialize SOEM
```

When investigating the error, the following solutions were found:

**1 -)** At this address: https://github.com/hans-robot/elfin_robot/issues/7

A solution is provided in [this comment](https://github.com/hans-robot/elfin_robot/issues/7#issuecomment-402581071)
"Hi, the Elfin should be connected directly to a computer, without the control box. In this case, you need such an adapter."
The mentioned adapter can be found here: https://github.com/hans-robot/elfin_robot/files/2164590/Elfin.cable.adapter.pdf

In addition to the relevant answer, a question was asked [here](https://github.com/hans-robot/elfin_robot/issues/31)
The response received [was](https://github.com/hans-robot/elfin_robot/issues/31#issuecomment-938323922) 

> 
> "Right now the only way to connect to the robot via ROS is to connect the LAN cable from the robot directly to your computer.
> The control box works mainly to power on (electrify) the robot.
> I recommend you buy a LAN to LAN adapter."

**Additional question:** Where and how should the EtherCat driver be installed? Is it necessary to install it?

Additionally, there is a mention of the Linux Real-Time kernel and Low-Latency kernel patching. The Low-Latency patch was attempted but successful results were not achieved. What are the differences between these two? Which one should be installed?

**Additional question:** Does the module inside the Control-Box that separates the EtherCat cable into power and cat cable perform the same function as the mentioned adapter? Can tests be conducted with that module until the adapter is obtained

Neverforgetlove, 2023-08-07:
When using the ROS controller, first ensure that the network cable is connected correctly and power on the robot in the teaching pendant. On your computer running ROS, please do not set a fixed IP address for the Ethercat network port and correctly set the network port name in the ROS config file.

question1：The installation steps of the EtherCat driver in the md file have been downloaded.

question2： Both real-time kernels and low-latency kernels aim to improve real-time performance. Real-time kernels are more focused on strict real-time task handling and predictability, while low-latency kernels prioritize system response speed and reduced execution latency while maintaining a certain level of flexibility and suitability for a wider range of application scenarios.

question3： If you can use the teaching pendant for power on operation, you can directly use it

When you have some problems with use real robots, you can contact our relevant technical support personnel, which will help you solve these problems faster.

oguzhanaknc, 2023-08-07:
> When using the ROS controller, first ensure that the network cable is connected correctly and power on the robot in the teaching pendant. On your computer running ROS, please do not set a fixed IP [...]

I am supplying power to the robot arm from the control box. I unplug the Ethernet cable between the robot arm and the control box and connect it to my computer. However, as soon as I disconnect the cable, the power to the robot arm gets cut off.

Neverforgetlove, 2023-08-07:
If you still cannot connect to ROS, I suggest you contact our after-sales personnel. We are happy to provide you with remote service and solve related problems

oguzhanaknc, 2023-08-07:
I thank you for your help. I have reached this stage now. But I am stuck again here.

```
Initializing etherCAT master
SOEM found and configured 4 slaves
SOEM IOMap size: 458
OPERATIONAL state not set, exiting
terminate called after throwing an instance of 'elfin_ethercat_driver::EtherCatError'
  what():  Could not initialize SOEM
```

Neverforgetlove, 2023-08-07:
I need to know the specific model and ROS version of your robot. It would be best to take a photo of the robot.

oguzhanaknc, 2023-08-07:
> I need to know the specific model and ROS version of your robot. It would be best to take a photo of the robot.

OS : Ubuntu 20.04
Ros : Noetic
Robot : Elfin5

Neverforgetlove, 2023-08-07:
What type of EndIO is it?Take a photo for me. And which branch of the ROS package are you using?

oguzhanaknc, 2023-08-07:
[image]

Neverforgetlove, 2023-08-07:
No this base ethercat, it's the end  of robot which on the sixth joint

oguzhanaknc, 2023-08-07:
[image]
 am currently conducting my tests through the control box attached to the card.

oguzhanaknc, 2023-08-07:
[image]
[image]

I am currently conducting my tests through the control box attached to the card.