huayan-robotics/elfin_robot #40: Difficulty Controlling DH Brand Gripper Attached to elfin5's endIO via ROS

oguzhanaknc, 2023-08-09:
I'm currently working on a project where I need to control a DH brand Gripper that is connected to the **endIO** of an **elfin5** robot using ROS. I have successfully managed to control the Gripper using the teaching pendant, but I'm facing challenges when it comes to controlling it through ROS.

I have searched extensively for documentation or resources on how to control this type of Gripper plugin using ROS, but unfortunately, I haven't been able to find any helpful information. It seems there's a lack of documentation specifically addressing the integration of the DH Gripper with ROS.

If anyone has experience with a similar setup or knows where I can find relevant documentation, tutorials, or examples, I would greatly appreciate your guidance. It would be incredibly helpful to have a resource that outlines the steps or provides code examples for controlling the Gripper using ROS commands.

Any assistance or direction on this matter would be highly valued. Thank you in advance for your help!

Neverforgetlove, 2023-08-11:
Currently, using ROS can only control the two EndIO, which is not yet supported for 485 protocol

oguzhanaknc, 2023-08-11:
Thank yoou for your attention, How can I control the gripper using just plain Python or C++ code without using ROS? What's the operation like inside the teaching pendant?

Neverforgetlove, 2023-08-11:
You can use the robot's SDK to control the end 485 to control gripper.This is a relatively complex process that is difficult to describe here. You can request explanations docs from our business department  in this regard

oguzhanaknc, 2023-08-11:
Can I control GIPPER without cotrolbox dependency?