huayan-robotics/elfin_robot #42: Error in Elfin 5 Moveit Configuration with Parellel Gripper

ritviksingh94, 2023-11-07:
Hi Team,
We are looking to integrate a custom made parallel gripper with an elfin 5 robot using ROS move it config. We have added the gripper (see description below) into the urdf. The gripper flange and the two fingers of the gripper have been added as 3 separate links. The joint between the gripper flange and elfin link6 has been defined as fixed, whereas the joints between fingers and gripper flange have been defined as prismatic. All these links and gripper have been added into the elfin 5 urdf. As far the structure of the srdf goes, the gripper flange (which is the end link of the robot) is added as the child link and since parent links are added automatically all the links of the robot except the gripper fingers are added in the first group “elfin_arm”. The second group “elfin_end_group” consists of the gripper finger links and the prismatic joints associated with them. The third group defined is ‘elfin_arm_parellel_gripper” which has the two groups “elfin_arm” and “elfin_end_group”. On running simulation in Rviz whenever a goal is assigned, the following error appears and the robot fails to move to the goal. 

**ERROR: Unable to get end effector tips from jmg.**

The same simulation runs smoothly when no gripper is integrated. Can someone help us with the general structure of urdf and srdf showing how grippers are typically integrated with elfin 5 moveit configuration? Or any leads as to why this error appears would also be greatly appreciated. 
[]([image])

Neverforgetlove, 2023-11-16:
I can't determine what happened from this picture,but we have recently integrated some grippers and currently preparing to update them. I can send you some packages to reference.

ritviksingh94, 2023-11-16:
> I can't determine what happened from this picture,but we have recently integrated some grippers and currently preparing to update them. I can send you some packages to reference.

Thank you, that would be really helpful. Could you share it whenever you get a chance? @Neverforgetlove

Neverforgetlove, 2023-11-16:
Yeah, I already know your email and I will send it to you later