huayan-robotics/elfin_robot #43: API for state feedback

Soil-L, 2023-12-13:
hello,
I am doing some control research using elfin3. I want to periodically get the state of joint(not state of encoder) and end. But i cannot find any api mentioned in api_description.md. 
Now i use some api of moveit to get states, but i think getting state from basic_api will be more convient.
So, i wandering that is there any API provided to get joint states and end states?

Neverforgetlove, 2023-12-18:
1. When you say joint status, do you mean the joint angle between the starting point and the end point of the trajectory?
2. Zero force functionality is not currently provided in ROS

Soil-L, 2023-12-18:
> 1. When you say joint status, do you mean the joint angle between the starting point and the end point of the trajectory?
> 2. Zero force functionality is not currently provided in ROS

1. Actually, i  just want the joint angles periodically whether it is moving or not. is it possible to use basic_api to achieve?
2. All right. is it in the develop plan?

Neverforgetlove, 2023-12-18:
1. In fact, you can see the joint radian position_tmp1 and position_tmp2 from the read_update function in elfin_ros_control/src/elfin_hardware_interface.cpp. You can customize a topic to publish it.
2. We are developing another control mode to implement. This mode will be quite different from the current ROS package and will not be completed as soon.

Soil-L, 2023-12-19:
> 1. In fact, you can see the joint radian position_tmp1 and position_tmp2 from the read_update function in elfin_ros_control/src/elfin_hardware_interface.cpp. You can customize a topic to publish it. [...]

1. OK, i am trying it.