huayan-robotics/elfin_robot #46: Problem with controlling E03

tina525241, 2024-04-16:
Hello, 

I'm using Ubuntu 20.04 Noetic and want to control E03 (485 end type) with ROS. 
However, when I followed the steps from the README.md file, the robot showed in Rviz had different pose from the real robot (all joint values were 0).
[image]
Besides, after the `roslaunch elfin_robot_bringup elfin_ros_control.launch` command, I got the following output:
--------------------------------------------------------------------------------------
process[elfin_ros_control-1]: started with pid [11434]
process[elfin_joint_controller_spawner-2]: started with pid [11435]
process[elfin_arm_controller_spawner-3]: started with pid [11436]
Initializing etherCAT master
SOEM found and configured 3 slaves
SOEM IOMap size: 384

Finished configuration successfully

 recognizing joint positions, please wait a few minutes ... ... 
Failed to read from ret:-2, slave_no:3, index:0x2033, subidx:0x00
The voltage of slave 1 is: 48.109125V.
[ WARN] [1713250622.002562999]: recognizePose phase1 failed while pose recognition in slave 1, channel 1
The voltage of slave 2 is: 48.390899V.
[ WARN] [1713250622.009947134]: recognizePose phase1 failed while pose recognition in slave 2, channel 1
[ INFO] [1713250622.010035864]: positions are recognized automatically
Loaded 'joint_state_controller'
Loaded 'elfin_arm_controller'
Started ['joint_state_controller'] successfully
[elfin_arm_controller_spawner-3] process has finished cleanly
log file: /root/.ros/log/4e9a8ff0-fbbe-11ee-b7fa-59996d2ba316/elfin_arm_controller_spawner-3*.log
[elfin_joint_controller_spawner-2] process has finished cleanly
log file: /root/.ros/log/4e9a8ff0-fbbe-11ee-b7fa-59996d2ba316/elfin_joint_controller_spawner-2*.log
-------------------------------------------------------------------------------

Looks like there's some problem with slave 3 and the recognizePose thingy. Any suggestions are appreciated. Thank you.

closed 2024-04-19 by tina525241

dilisharanasinghe, 2024-10-18:
How did you fix the issue?

tina525241, 2024-10-18:
The incorrect pose in RViz was due to the incorrect `count_zeros` in config/elfin_drivers.yaml file. `Failed to read slave...` is an ignorable message in my case.

dilisharanasinghe, 2024-10-18:
Thank you, setting `count_zeros` worked. mine - `count_zeros: [6706834, 1966878, 6039704, 2370590, 5667893, 5755803]`