huayan-robotics/elfin_robot #47: 你好关于Gazebo相关问题

Panda12580, 2024-07-03:
请问一下为什么我将urdf添加到gazebo中后，它会自己慢慢的移动，这大概是由什么原因造成的？

Neverforgetlove, 2024-07-11:
gazebo中有物理仿真参数，比如摩擦力、重力等等，这与你的模型参数和仿真配置有关

Panda12580, 2024-07-11:
不好意思，我还是个新手。
我是通过读取urdf生成gazebo的模型，都是默认的为什么会这样，还有一个问题就是您提供的gazebo的包他并没有设置控制器的pid，该如何解决呢。

Neverforgetlove, 2024-07-12:
你的模型是什么模型？ elfin包中的控制器默认为位置控制模式，不需要PID

Panda12580, 2024-07-12:
我git clone你的包后能够编译通过，但是我用roslaunch elfin_gazebo elfin3_empty_world.launch，会报错显示没有设置P参数定义
[image]

Neverforgetlove, 2024-07-12:
这个报错不影响gazebo中的仿真使用

OneCat11, 2024-11-11:
请问运行 roslaunch elfin_gazebo elfin3_empty_world.launch  直接报错是啥原因啊
!
[image]

Panda12580, 2024-11-13:
> 请问运行 roslaunch elfin_gazebo elfin3_empty_world.launch 直接报错是啥原因啊 ! ![微信图片_20241111192011](https://private-user-images.githubusercontent.com/54233635/384900825-a8218086-8f08-40a3-a3e1-0acbb4772c08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NjY2NjAsIm5iZiI6MTczMTQ2NjM2MCwicGF0aCI6Ii81NDIzMzYzNS8zODQ5MDA4MjUtYTgyMTgwODYtOGYwOC00MGEzLWEzZTEtMGFjYmI0NzcyYzA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDAyNTI0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhZDM1MTBlYTNhNzdmNGVlM2FjMDQ5YWM3OGQxOTZiOGRkNzc0MzY3OGRlNjYxYTQ4NWJiYWE4YTk5MjE2YzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TMkTHtjsPsX4D6oYDHWGjWetEv1L5x92-Y3Tpufd5dE)

你没有截全，你urdf_spawer报错上面还有没有问题，正常来说应该是你的缺东西或者环境不对，我下他的包编译后是直接能跑的

Panda12580, 2024-11-13:
> 这个报错不影响gazebo中的仿真使用

谢谢您的答复！

OneCat11, 2024-11-14:
> > 请问运行roslaunch elfin_gazebo elfin3_empty_world.launch直接报错是什么原因啊！![微信图片_20241111192011](https://private-user-images.githubusercontent.com/54233635/384900825-a8218086-8f08-40a3-a3e1-0acbb4772c08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE0NjY2NjAsIm5iZiI6MTczMTQ2NjM2MCwicGF0aCI6Ii81NDIzMzYzNS8zODQ5MDA4MjUtYTgyMTgwODYtOGYwOC00MGEzLWEzZTEtMGFjYmI0NzcyYzA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEzVDAyNTI0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhZDM1MTBlYTNhNzdmNGVlM2FjMDQ5YWM3OGQxOTZiOGRkNzc0MzY3OGRlNjYxYTQ4NWJiYWE4YTk5MjE2YzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TMkTHtjsPsX4D6oYDHWGjWetEv1L5x92-Y3Tpufd5dE)
> 
> 你没有截全，你的urdf_spawer报错上面还有没有问题，正常来说应该是你的缺东西或者环境不对，我下他的包编译后是直接能运行的

这是18.04系统上面没有问题，我换了20.04系统后就可以打开运行了，虽然依旧有报错，但是不妨碍仿真