huayan-robotics/elfin_robot #48: Cannot enable servo

vaibhavwalia921, 2024-09-10:
I did a fresh installation of ubuntu 20 and ros noetic. On that pc i cloned the package elfin_robot noetic_ethercat branch. It is built normally without any error. I launch the pipeline as the following:
Elfin_robot_bringup.launch
Elfin_ros_control.launch
Moveit_planning_execrution. Launch

But when i launch elfin_api.launch
I was getting the error no module found wx

So i installed this using 
Pip install wxPython

And the API is running and i can see the GUI. 

The problem is when i do servo enable. I hear brakes are disabled and then enabled again. 

In the terminal i get this error: 
Robot servo not enabled.

Note: previously everything was working. I started to get this problem on fresh installation of ubuntu.

Neverforgetlove, 2024-09-11:
According to the current information, except for the high voltage, everything seems normal from the PDO information. Could this robot be used normally before? If the voltage can be reduced to about 48V, and the enable still fails, read txpdo again, and use wireshark to capture the ethercat network port, click the 'servo on' button again, and send the recorded data.

vaibhavwalia921, 2024-09-11:
Thanks for the response. 

Yes, the robot was working absolutely fine. I have attached a gripper to it and configured moveit_config package to use gripper and arm together. Everything was working normally. The problem came after the fresh  a fresh installation of ubuntu 20.04. The voltage is high because the robot is connected through a battery And it’s just overcharged but it worked normally in these conditions as well. 

My first assumption was it might be because of the PREEMPT_RT patch. So I tested it with both the generic and the patched kernel. Still the problem is the same. However, we were using it before without PREEMPT_RT patch and it used to work.

Neverforgetlove, 2024-09-12:
Is the gripper install at the end of the robot? I think the kernel does not affect the enable，the enabling process is determined by the control word and status word, when you read the txpdo, read the rxpdo information at the same time.

Astexia-Johnny, 2025-02-19:
无法开启使能。
你好，我这里用ubuntu20.04 和ros noetic ，elfin5机械臂。
按照技术方案进行环境的配置与部署。
机械臂前期是稳定运行过一段时间，也可以正常使能servo on，控制机械臂移动，今天我按顺序运行

$ roslaunch elfin_robot_bringup elfin5_bringup.launch

$ sudo chrt 10 bash
$ roslaunch elfin_robot_bringup elfin_ros_control.launch

$ roslaunch elfin5_moveit_config moveit_planning_execution.launch

$ roslaunch elfin_basic_api elfin_basic_api.launch

**现象：**在我点击servo on时候，机械臂在使能开启的下一刻，马上自动关闭了使能，并出现以下报错。

[image]

api窗口报错如下
Failed to start the default controller

[image]

以下是txpdo
rosservice call /elfin_ros_control/elfin/get_txpdo "data: true"
success: True
message: "slave1_txpdo:\n0x21:0x00:0xf8:0xff:0xf1:0x18:0xf3:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x0f:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x21:0x00:0x00:0x00:0x9a:0x9b:0xac:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:\n\
  slave2_txpdo:\n0x21:0x00:0x17:0x00:0x13:0x6a:0x56:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x0f:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x21:0x00:0xf4:0xff:0x01:0x51:0x02:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:\n\
  slave3_txpdo:\n0x21:0x00:0xfe:0x00:0x07:0xc4:0x7d:0x01:0xff:0xff:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x0f:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x21:0x00:0x11:0x00:0xc4:0x34:0xaa:0x01:0xff:0xff:0x00:0x00:0x00:0x00:0x08:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:0x00:\n"

在此之前，机械臂并未有任何硬件方面的变动，并未有改装或加装东西，突然就这样了。
我也单独用示教器控制机械臂，能正常开启使能并控制机械臂移动。
希望能得到您的帮助。