huayan-robotics/elfin_robot #49: Is elfin full support ros2?

goddard73, 2024-09-24:
I just started working with a robotic arm and I would like to know if Elfin fully supports Ros2 to choose the way I want to operate the robotic arm, Thank you.

goddard73, 2024-09-25:
I have one Elfin 5 and one Elfin 5-L, I want to use for robotic grasping.

I am wondering is that could use ros2.

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/25/2024 15:31 |
| To | hans-robot/elfin_robot ***@***.***> |
| Cc | goddard73 ***@***.***>,
Author ***@***.***> |
| Subject | Re: [hans-robot/elfin_robot] Is elfin full support ros2? (Issue #49) |

Elfin-E series supports foxy and humble now, whitch way do you want to operate the robotic arm?

Neverforgetlove, 2024-09-25:
Sure,we have ros2 branch,you can find it in elfin_robot_ros2

goddard73, 2024-09-25:
can you give me a link, I have not found where to download, thank you.

---- Replied Message ----
| From | ***@***.***> |
| Date | 09/25/2024 16:44 |
| To | hans-robot/elfin_robot ***@***.***> |
| Cc | goddard73 ***@***.***>,
Author ***@***.***> |
| Subject | Re: [hans-robot/elfin_robot] Is elfin full support ros2? (Issue #49) |

Sure,we have ros2 branch,you can find it in elfin_robot_ros2

Neverforgetlove, 2024-09-25:
https://github.com/hans-robot/elfin_robot_ros2