huayan-robotics/elfin_robot #50: 关于我在使用ROS控制elfin3遇到的报错

OneCat11, 2024-12-09:
首先我的系统为20.04
运行roslaunch elfin_robot_bringup elfin3_bringup.launch命令
报错：... logging to /home/bu/.ros/log/f55ec08a-b3ac-11ef-819f-4bc9cffd9016/roslaunch-bu-G3-3579-18190.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://bu-G3-3579:40735/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.17.0

NODES
  /
    robot_state_publisher (robot_state_publisher/state_publisher)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [robot_state_publisher/state_publisher]: Cannot locate node of type [state_publisher] in package [robot_state_publisher]. Make sure file exists in package path and permission is set to executable (chmod +x)
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
通过修改elfin3_bringup.launch文件，将里面的type="state_publisher"修改为type="robot_state_publisher"之后能够成功打开RVIZ界面。
-------------------------------------------------------------------------------------------
以及在RVIZ界面里规划运行机械臂时终端报错如图：
[image]
求助up，我的错误该如何解决才能控制机器人

Neverforgetlove, 2024-12-17:
如果是连接真实机器人，需要运行roslaunch elfin_robot_bringup elfin_ros_control.launch成功后再启动moveit和rviz,如果运行的是gazebo仿真就需要启动gazebo仿真

OneCat11, 2024-12-17:
> 如果是连接真实机器人，需要运行roslaunch elfin_robot_bringup elfin_ros_control.launch成功后再启动moveit和rviz,如果运行的是gazebo仿真就需要启动gazebo仿真

谢谢，已解决，已经实现通讯了